km4ack/pi-build #398: Bullseye - Pi4 - FlRig issues

thufer2002, 2022-04-10:
Flrig works for about 2 contacts then locks up in transmit but with no signal driving the xmtr. Can't reboot PI4 must power off/on to reset. Flrig will take a long time to close if "X'ing out of app. As noted by another user Flrig open with 20M selected, also the apps location on the desktop is not remembered and always starts at the same position no mater what I do.

My setup:
Pi4 with Flirc case, UBlock GPS, IC-705 Rig

km4ack, 2022-04-11:
I could be wrong but this sounds more like an RFI issue than an FLRIG issue. I have the latest version of FLRIG running with my 705 without any issues. Try shortening the USB cable between the Pi and the 705 and adding a ferrite bead or two. 

Another test is to reduce to power of the 705 to 1/2 watt and see if the same issue occurs. If not, it is almost certainly RFI getting into the Pi.

thufer2002, 2022-04-11:
Some suggestions for ferrite beads? Amazon link? I’ll try it.

 

Though the same length of cable and a PI Zero w2 (Non Bullseye) works just fine. Both pi’s built within last 3 days

closed 2022-08-26 by km4ack