km4ack/pi-build #400: RepeaterSTART

programmin1, 2022-04-22:
Good morning, I was wondering if you might include RepeaterSTART for Linux, the Linux repeaters app I've been working on -

https://github.com/programmin1/Repeater-START

km4ack, 2022-07-10:
Love the idea and the app! I would like to do a video on it soon. Is there a way to manually pull in the latest repeater info while running the app on a Pi?

programmin1, 2022-07-10:
Thanks. When connected, the app pulls the repeater data upon opening the app and every 10min while open. On the Android version there is an option for the update interval. Did you see the contest for anyone interested in improving the listings in the repeater listing? - https://hearham.com/contest2022

km4ack, 2022-07-10:
I did see the contest and will promote it in my newsletter next week. I'll also mention it in the video. I submitted one of our local repeaters this afternoon through hearham. Once I know I did it correctly, I'll add another during the video to show the process. I am going to shoot for the video to release on the 19th or 26th of July 2022.

programmin1, 2022-07-10:
Whose newsletter?
I should note that repeaters are added to the database and visible for everyone, only after someone goes in and approves it.

km4ack, 2022-07-11:
The newsletter is one that I put out each week that has ~9000 subscribers. It goes out each Tuesday morning at 8am. Here's the copy that will go out on July 12:

Another project I want to get on your radar is [RepeaterSTART](https://github.com/programmin1/Repeater-START). It's a cool project that gives you an offline directory of repeaters. However, the developer needs our help adding repeaters to the database and is running a [contest this summer](https://hearham.com/contest2022). If each reader added just one repeater to the list, we could increase the database by almost 10,000 repeaters! I'll be doing a video on how to get it installed on the Pi in a couple of weeks and the app will be added to Build a Pi in the near future. For now, please head over to [the site](https://hearham.com/contribute) and add one repeater from your area. Oh, and for the record, the app is also available for Windows and Android.

km4ack, 2022-07-11:
I was working on the upcoming video and noticed a mistake in the repeater that I submitted. The repeater call sign is KU4B. The database has it listed as 175.170 but the correct frequency is 145.170. I apologize if I fat fingered it. Didn't see a way on the website to submit a correction.

programmin1, 2022-07-11:
> I was working on the upcoming video and noticed a mistake in the repeater that I submitted. The repeater call sign is KU4B. The database has it listed as 175.170 but the correct frequency is 145.170. I apologize if I fat fingered it. Didn't see a way on the website to submit a correction.

No problem, there is no  edit or "change" button only "Add a comment" if you scroll to the very bottom.

km4ack, 2022-10-19:
issue resolved in 3.3.0

closed 2022-10-19 by km4ack