km4ack/pi-build #405: Conky is not starting  after new install bap

RonMar26, 2022-06-06:
I did a few times a new bap install and every time conky is not starting. Also not when I start it out of the menu.
It seems to be no gps problem. Cgps shows all the info. Disconnecting gps device is also not showing up conky.
The pi os I use is bullseye 32 bit.
What can be my problem

km4ack, 2022-06-06:
from the terminal, run `conky` and post any errors please.

RonMar26, 2022-06-06:
pi@raspberrypi:~ $ conky
conky: desktop window (1c00010) is subwindow of root window (3b0)
conky: window type - normal
conky: drawing to created window (0x1a00001)
conky: drawing to double buffer
conky: forked to background, pid is 1809
pi@raspberrypi:~ $ 
cat: /run/user/1000/gridinfo.txt: Bestand of map bestaat niet
conky: $tail can't find information about /home/pi/Documents/mylog.txt

km4ack, 2022-06-06:
Is your user name "pi"? BAP currently only works when the user name is Pi. There is a beta version being tested that removes this requirement.

km4ack, 2022-06-06:
ok. try running:
`touch /run/user/1000/gridinfo.txt`
and then
`mkdir -p /home/pi/Documents`
`touch /home/pi/Documents/mylog.txt`
Finally, try launching conky again with 
`conky`

RonMar26, 2022-06-06:
pi@raspberrypi:~ $ conky
conky: desktop window (1a00010) is subwindow of root window (3b0)
conky: window type - normal
conky: drawing to created window (0x1e00001)
conky: drawing to double buffer
conky: forked to background, pid is 4683
pi@raspberrypi:~ $ 
conky: $tail can't find information about /home/pi/Documents/mylog.txt

This is now the message after I have tried to start conky in the menu

RonMar26, 2022-06-06:
sorry i go to try that

RonMar26, 2022-06-06:
Yes now it is working again in a small version. But that can be changed.

TNX Jason all is working

RonMar26, 2022-06-06:
Also after reboot is conky working again.
Tnx

73 PD7RON
Ron

closed 2022-06-06 by km4ack