km4ack/pi-build #422: QSSTV Site Down

km4ack, 2022-09-16:
The [site](http://users.telenet.be/on4qz/qsstv/downloads/) where BAP downloads QSSTV is down. Thanks Bruce, W1BAW, for catching the issue.

km4ack, 2022-09-16:
A bit more digging netted the following info:

> 
> Do you have a Telenet (Business) internet subscription and your website is no longer available? This is because we took the web space that is included with or linked to your Telenet (Business) internet offline on September 1, 2022. We did this because there are now many new, more secure and more advanced options for managing web space.
> 
> 
> We previously informed you about the termination by e-mail or by letter and via your statement or invoice. Missed info? You can read everything about the termination of your web space on our Customer Service. You will also discover how to download your old website files before they disappear permanently and what alternative solutions there are.

The new site needs to be located or QSSTV needs to be removed from BAP until this is resolved.

SpudGunMan, 2022-09-16:
Things over there are spicy. I think a copy is on a package might need to resort to for now. I do think archives should be grabbed just in case. 

I hope we hear from him soon and the site is back - all the best

km4ack, 2022-09-16:
> Things over there are spicy. I think a copy is on a package might need to resort to for now. I do think archives should be grabbed just in case.
> 
> I hope we hear from him soon and the site is back - all the best

Do you have a link where I can download the latest source files?

SpudGunMan, 2022-09-16:
here is from archive.org https://web.archive.org/web/20220811214139/http://users.telenet.be/on4qz/
also I noticed people started forking to git one example of.. https://github.com/KD9QZO/qsstv

I haven't had email back from the developer for a bit he was responsive about 3 years back

SpudGunMan, 2022-09-16:
I also noticed fedora package has archives https://src.fedoraproject.org/repo/pkgs/qsstv/

km4ack, 2022-09-20:
Another possible source - https://launchpad.net/ubuntu/+source/qsstv

SpudGunMan, 2022-10-05:
Might have found it, search engines removed all the old domain at this point. 

https://github.com/ON4QZ/QSSTV

Haven't heard from OM via email but has active development on it under same call

SpudGunMan, 2022-10-11:
suggested patch https://github.com/km4ack/pi-build/pull/433

km4ack, 2022-10-19:
Resolved in https://github.com/km4ack/pi-build/pull/444

closed 2023-02-05 by km4ack