km4ack/pi-build #423: Build A Pi packet not working

radiohamming, 2022-09-18:
I installed Build-A-Pi, I followed all the directions,  and I watched all of the videos.

I was able to telnet messages but for the life of me, I cannot get Packet to work. I get Dial timeout messages, and unable to establish connection messages! I'm using a Tigertronics Signalink and I also tried a Sabrent sound card. I clicked on the direwolf icon from the menu and a box opens-up and I can watch the text flow so I know that the sound cards are working. What could possibly be wrong, I just can't transmit, the modem is trying to transmit but it just gives me timeouts, and unable to establish connections?

I'm using the latest 32-bit Pi OS and the latest Build-A-Pi.
I have seen posts on the internet that others are having the same problems, It's not just me.

Ken Hudson

SpudGunMan, 2022-09-18:
You need a terminal or app to access the tnc (direwolf) 

Via agw port or kiss port not telnet

radiohamming, 2022-09-18:
Direwolf and Kiss are installed into Build-A-Pi. Both are illuminated green
to show they have started on the Crony. Pat web page opens up, I can telnet
but I can’t transmit packets! I know what telnet is, I want to transmit
over the radio. Something else is wrong.

Ken H KL2HF

SpudGunMan, 2022-09-18:
So your trying to connect to winlink rms stations?

You would either telnet to winlink or use the radio you would not be using telnet with a radio

radiohamming, 2022-09-18:
I used Telnet to test the system to see if anything was working. Telnet worked. Now I’m trying to get the radio portion to work.

km4ack, 2022-09-18:
Choosing direwolf from the main menu isn't the best approach unless you want to start the other needed components manually from the command line. Instead, open Pat Menu->Start/Stop Modem->Start Packet Modem. The reason is that you don't just need direwolf running. You also need kissattach, kissattach parameters, and possibly rig control. Pat Menu will start all of these in the correct order when you use the start/stop modems feature. Also, be sure that you update Pat Menu settings when you open it the first time. Most of the defaults will work but you will need to choose rig control if that is something you intend on using. Did you configure FLRIG for your radio? It is needed if you intend to use rig control. With the signalink, rig control would be optional since the Signalink can handle PTT through the its built in VOX circuit. If you want to use the sabrent sound card, you will need to configure rig control since this card relies on methods other than VOX to key the radio. Should you have issues with the Signalink keying the radio, search the forum. A common problem is volume on the pi. It has been covered many times on the forum as in [this post](https://groups.io/g/KM4ACK-Pi/topic/93476289#8837).

Finally, this isn't a bug in Build a Pi. As such, it would be better to ask questions in [the forum](https://groups.io/g/KM4ACK-Pi/topics) where lots of people visit each day and can help in making your first connection.

closed 2022-09-18 by km4ack