km4ack/pi-build #447: gridtracker not installing :: How to fix manually

km4ack, 2022-10-17:
Gridtracker is currently not installing. Until this is resolved in BAP, use the commands below to install. These directions are for the 32 bit system. If you are running the 64bit Pi OS, substitute arm64 for arm32. The process takes ~5 minutes. The first command can be skipped if you haven't previously installed Gridtracker.
```
rm -rf ~/GridTracker
cd ~/Downloads
wget https://downloads.gridtracker.org/v1.22.1010/GridTracker-1.22.1010-linux-arm32.tar.gz
tar -xvzf GridTracker-1.22.1010-linux-arm32.tar.gz
rm GridTracker-1.22.1010-linux-arm32.tar.gz
mv GridTracker-1.22.1010-linux-arm32/ ~/GridTracker
```

SpudGunMan, 2022-10-17:
you can likley scrape https://gridtracker.org/release.html with better success

km4ack, 2022-10-17:
I'm done dealing with this issue. They have moved the downloads numerous times. I created [a file](https://github.com/km4ack/pi-scripts/blob/master/gt-download-links) outside of BAP that can be updated by me on a whim with the latest download links. This has been a PIA for almost 2 years now. The new file will allow me to update the links without having to update a single thing in BAP. Reference the above commit 98a01e7

SpudGunMan, 2022-10-17:
Haha I agree that page is always moving around! 

Yea reasonable way to handle it, you ever use python to scrape web with you can reference the div and it's a lot more program friendly vs scraping curl on complex pages.

km4ack, 2022-10-19:
issue resolved in 3.3.0

commit 98a01e72 (2022-10-17): fix #447
closed 2022-10-19 by km4ack