km4ack/pi-build #451: chirp pi4 64bit install issue

SpudGunMan, 2022-10-17:
on install of new 64bit pi4 Chirp icon is there but points to missing file

SpudGunMan, 2022-10-17:
https://github.com/goldstar611/chirp-appimage/releases has packages for 64/32 ARM

SpudGunMan, 2022-10-17:
working on a logic check here https://github.com/SpudGunMan/pi-build/tree/chirp-fix

km4ack, 2022-10-19:
Let's see if we can also update the app-check script to find out if the latest version of this is installed and give the user better feedback. See [this section of code](https://github.com/km4ack/pi-build/blob/master/app-check#L241-L249).

SpudGunMan, 2022-10-19:
> Let's see if we can also update the app-check script to find out if the latest version of this is installed and give the user better feedback. See [this section of code](https://github.com/km4ack/pi-build/blob/master/app-check#L241-L249).

looking into it

SpudGunMan, 2022-10-19:
updated with https://github.com/km4ack/pi-build/pull/459

km4ack, 2022-10-19:
[PR 459](https://github.com/km4ack/pi-build/pull/459) merged into dev branch