km4ack/pi-build #454: ARDROP ARM64 issues

SpudGunMan, 2022-10-18:
there is no precompiled binary for ARM32
source of code/ packages 
https://www.cantab.net/users/john.wiseman/Downloads/Beta/

SpudGunMan, 2022-10-18:
this modem works fine in BAP/pi3 and pi4/32 there is not currently a build for arm64 on the dev website. but I was pondering a quick fix and this is a possible option.

mkdir ~/.wine/drive_c/ardopc/
wget https://www.cantab.net/users/john.wiseman/Downloads/Beta/ARDOPC.exe
env WINEDEBUG=-all wine /home/pi/.wine/drive_c/ardopc/ARDOPC.exe

as this issue is only a pi4/64 bit issue and the issue can be resolved with emulation it might be the quickest fix?

km4ack, 2022-10-18:
No need to bring WINE into the mix. We simply install the needed 32bit libraries even if running a 64bit machine. See [this section](https://github.com/km4ack/pi-build/blob/master/functions/base.function#L139-L143) of code

closed 2022-10-18 by km4ack