km4ack/pi-build #475: issue notepad.exe from wine

SpudGunMan, 2022-10-20:
should remove this entirely from the OG wine script its self or clean it up here

remove the notepad as a file to open filetype .txt

SpudGunMan, 2022-10-20:
tag @WheezyE this work I am not hip enough to tag in git https://github.com/WheezyE

SpudGunMan, 2022-10-20:
I think this is default wine install package setting this up

SpudGunMan, 2022-10-21:
~/.local/share/mime

might be the spot to mess about

WheezyE, 2022-10-21:
Are you wanting to remove the default notepad.exe from Wine? I think Windows programs sometimes launch notepad when a users click on a readme button or a view log button. It might be nice to have and I think doesn’t take up much space

SpudGunMan, 2022-10-21:
no its good to have notepad in wine install for windows errors that would happen.. 
however its not ideal to click on a .txt file in raspberry land and get notepad.exe as your editor due to emulation tax

so editing the files with bash after install during clean up in the ~/.local/share/mime

folder is a file for registering the "send to" " open with" data for the user

I think this behavior might be default to wine which shows off its coolness but .. only cool when CPU power is limitless

hope that helps? I can come back later and drop some code for this as well just wanted to get the idea down to review it