km4ack/pi-build #484: Build-A-Pi on Quadra Inovato

KK4ECR, 2022-11-14:
What would it take to run Build-A-Pi on the Quadra Inovato.  It is a much less expensive option to the raspberry pi.  I love the options that Build-A-Pi offers and want to run it on my Inovato.

Thank!

yv6eda, 2023-01-04:
Same here! No answer yet?

Jorge VE3EAD

SpudGunMan, 2023-01-04:
73 linux bapp files should work

I have tested on a fork with success.

siskiyous, 2023-01-10:
I too hope you are able to create a script that will run on the QI, or maybe some day Mint. Thanks for your hard work Jason. - WS7PB 73

DanDawson, 2023-01-22:
Has any one even attempted Build-a-Pi on a Quadra Inovato? Just looking at [their site](https://www.inovato.com/) they claim:

> **Run most Linux and Raspberry Pi apps**
> Quadra is broadly compatible with 64-bit Raspberry Pi software. It includes the [Pi-Apps installer](https://github.com/Botspot/pi-apps) to make installation of many popular apps easy.

So if any of you have one... would you be willing to at least give it a try and report back? 

In the same way, I wonder if it would run on a [Libre Computer Board AML-S905X-CC (Le Potato) 2GB 64-bit Mini Computer for 4K Media](https://amzn.to/3J7JpYq). 

Since RPi4's have been unavailable for so long, I'm looking for alternatives.

SpudGunMan, 2023-01-22:
Yea it mostly works fine it's not raspan direct so few hiccups. 

If you find some report them!

DanDawson, 2023-01-22:
Thanks for the update @SpudGunMan, I appreciate it! If I end up giving that route a try I'll report back on my findings.