km4ack/pi-build #486: Pat Setup with KPC 3+.

apelouze, 2022-11-29:
I want to configure Pat to work with a KPC 3+ to FT-8800 via data port.  Telnet connections works; I have Pat and AX25 installed.

I have played with the setting to no avail.  Can I be pointed to a tutorial walking through hardware TNC configuration?

Thanks for the help!

Aaron KD6FLM

SpudGunMan, 2022-11-29:
you might be missing rigctl/hamlib https://github.com/la5nta/pat/wiki/Rig-control

apelouze, 2022-11-29:
Rig control is not an option on that radio.  I install the package to see, no change.

I have Rig control set to No in Pat settings; in the Pat Connect to remote node page, the freq has a red line through the box.  I am guessing that the software still thinks it has control of the radio.

SpudGunMan, 2022-11-29:
I missed the KPC3+ hardware TNC comment for some reason .. you need to fix that device to control your PTT
https://www.saresrg.org/Kpc3+_ft-7900-8800_tnc_cable.txt

apelouze, 2022-11-29:
I have the cable wired per the diagram; I can connect to local packet stations.

SpudGunMan, 2022-11-29:
you likely want to ask in the PAT user areas,  https://groups.google.com/g/pat-users/c/ox3mCZI7qGA check out this thread on topic.