km4ack/pi-build #488: 32 bit linbpq doesn't run on (some) 64 bit Raspbian

scottredd, 2022-12-22:
Hello!

First, thanks for the great work on Build-A-Pi. It's really nice to be able to get devices up and running so easily.

It seems that in the week or so that new installs of linbpq won't run on my 64 bit Raspbian. When trying to launch linbpq, or the configuration utility, I get a "file not found" error, despite it being right there. I think it might have something to do with the 32 bit vs. 64 bit architecture.

The weird thing is I have the same Raspbian image on two identical boards, but the one built last week will run linbpq and the one built today will not. Even if I copy the binary from the old pi to the new pi, it still won't run on the newer pi.

Are there any tricks to getting 32 bit linbpq to run on 64 bit OS? Or can anyone think of something that I might have done (by accident) to make the old pi work that I'm forgetting on the new one?

Thanks!

Scott
W3RED

scottredd, 2022-12-22:
It's worth noting that I am using a Libre Computer Le Potato 2G (AML-S905X-CC).

SpudGunMan, 2022-12-22:
Lots of support on groups.io bpq group as well. 

For pi you need to have arm 386 packages loaded as well, which the script should be loading. 

Centric around raspian and the pi, what Os you running?

scottredd, 2022-12-22:
Yes, I though the build-a-pi script handled all the dependencies. It's just so weird, because I always just run the script and let it do its magic. I can't think of anything I did differently on the older install.

I did do an update and upgrade on the new system. I'm hesitant to do that on the old system since it's running my digipeater and PBBS.

Here's the uname -a from both systems:

New (doesn't work):
Linux raspberrypi 6.0.15-01069-g12e53ea7f114 #1 SMP PREEMPT_DYNAMIC Thu Dec 22 01:18:46 UTC 2022 aarch64 GNU/Linux
p

Old (does work):
Linux digi2 6.0.12-00858-gb98721ea4575 #1 SMP PREEMPT_DYNAMIC Thu Dec 8 19:08:19 UTC 2022 aarch64 GNU/Linux

SpudGunMan, 2022-12-23:
http://www.cantab.net/users/john.wiseman/Downloads/Beta/pilinbpq

just downloaded and seems to be fine, I would move this to the BPQ groups.io as its really more that project?

scottredd, 2022-12-27:
Thanks for the replies. I went over the to BPQ forum and started a discussion. After some back and forth, I got it working. Here are some notes in case someone else is having the same problem:

**SOLVED**

After I copied Raspbian to my SD card, then I created my default user, SSH flag, and WPA Supplicant. Then I power up the Le Potato and I ssh in to run the following:

sudo dpkg --add-architecture armhf
sudo apt update
sudo apt upgrade -y
sudo apt install xrdp -y
sudo apt install libc6:armhf -y
sudo apt install libasound2:armhf -y
sudo apt install zlib1g:armhf -y
sudo apt install libqt5widgets5:armhf -y

After a reboot, I RDP in to the system, open a command shell, and run the following to install the desired software using Build-A-Pi.

git clone https://github.com/km4ack/pi-build.git $HOME/pi-build && bash $HOME/pi-build/build-a-pi

After a reboot, I log in again, change to the linbpq directory and start up linbpq. Works like a charm!

Thanks much for the help.

Scott
W3RED

closed 2022-12-27 by scottredd