km4ack/pi-build #489: WSJTX won't download

rexhunt, 2023-01-03:
The WSJTX installation can't download it's files as the physics.princeton site is down.

SpudGunMan, 2023-01-17:
Here is a possible fix

```

FT8PKG=$(curl -s https://wsjt.sourceforge.io/wsjtx.html| grep .tgz | sed 's/.*="//;s/">.*$//' | sed 's/^.*files\///' | awk '{split($0,a,"/"); print a[2]}')
FT8DIR=$(curl -s https://wsjt.sourceforge.io/wsjtx.html| grep .tgz | sed 's/.*="//;s/">.*$//' | sed 's/^.*files\///' | awk '{split($0,a,"/"); print a[1]}')

cd ${BUILDDIR}
wget --tries 2 --connect-timeout=20 $(curl -s https://wsjt.sourceforge.io/wsjtx.html | grep .tgz | sed 's/.*="//;s/">.*$//' | head -1) -O $FT8PKG

```

michaelShriver, 2023-01-17:
Nice timing, I was just working on a similar patch. See my pull request :)

commit cfce232f (2023-02-05): fix #489
closed 2023-02-05 by km4ack