km4ack/pi-build #501: Conky

MPlocar, 2023-02-12:
Unable to get the Conky screen to move to the top/right of the monitor. Screen res is 1920x1080

DanDawson, 2023-03-09:
If you're willing, give this a try and report back if it takes care of your issue:

> sudo vi /boot/config.txt

Uncomment these two lines and adjust the resolution to 1920x1080:

> framebuffer_width=1920
> framebuffer_height=1080

Then finally:

> sudo reboot

k1ypp, 2024-02-24:
Hi Jason: just today I watched a video of yours and it convinced me I had to try your Pi-buld. It is now up and running and I need to configure a bunch of stuff. One thing you showed in a video was how to change the size of conky, it was right near the end of the video. I will be darned if I can find it.
My conky runs off the bottom of the RPi screen. Can you point me to that video? Thanks.
de Dennis, K1YPP