km4ack/pi-build #504: NMEA and PPS different with build a pi

mikey1963, 2023-03-10:
Without this build my pps time signal works fine but when build a pi is installed my pps signal just times out and is therefore unusable.  Also my NMEA output seems to be significantly slower than internet connected devices.  Without build a pi my gps whoops internet connected devices.
Having scanned thru the code I can't see anything affecting it, anyone else throw any light on this?
[image]

km4ack, 2023-03-10:
There is only one section in the code where BAP modifies things pertaining to the GPS.

https://github.com/km4ack/pi-build/blob/master/functions/base.function#L74-L85

You might compare /etc/default/gpsd.org to the current gpsd file to see if any of the BAP changes are causing this.

mikey1963, 2023-03-10:
Thanks I had already found this and like you , I can’t see anything in there to effect it.  I think the only other thing to try is to not use shared memory segments.  Perhaps use some other directly device addressing line in the chrome conf file I guess.  I will keep investigating.
Thanks for a great script.
Mike de M0JUJ

[http://static.ideasunplugged.com/signature/s_052/t_7edvEm.jpg?v=45]
Kind regards, Michael A Smith.

📲 07973 221971
🏠 24, Fifth Avenue, Portsmouth PO6 3PE UK
📩 ***@***.***
My apologies for any incorrect information or typographical errors. Any opinions expressed are my own, and are not intended to offend. Anyone offended by anything stated, will likely have offended me, by being inconsiderate of my beliefs.

https://github.com/km4ack/pi-build/blob/master/functions/base.function#L74-L85