km4ack/pi-build #505: JTDX and SDR Software

KC9RFV, 2023-03-20:
I love this software!! Was working off and on trying to get JS8CALL installed on my raspberry pi 3B for 2 weeks. Nothing I seemed to do worked... Your software installed first try!! Great Job!! Everything works as it should. I am so happy I have this software now. 

Was wondering about a couple of things:

Do you have plans on adding JTDX? I like this program better than WSJTX for some reason. Can't put my finger on it, but it just seems cleaner to me. Way easier to read than WSJTX. Was curious if you had any plans on adding it...

Also, any plans on adding any SDR software? GQRX, SDR++,  or CubicSDR? I have GQRX installed, but am struggling with CubicSDR and SDR++. Just curious if you had any plans on adding any of these.

Also, I am curious if this works on a 64 bit system as well?

Again, great job on an exceedingly great piece of software!!!

km4ack, 2023-03-20:
No current plans to implement SDR software but it can always be added on top of BAP. 

For a long time [I was blocked](https://github.com/km4ack/pi-build/issues/153) from the JTDX site. I just check and it seems the original site no longer exists. Where is the main site for JTDX?

KC9RFV, 2023-03-21:
I am not sure of the exact web site, but I was able to get the code using the following command. It finally compiled and installed for me. Took me forever to get everything installed that I needed. I am still learning linux and raspberry pi so I can't give you specifics. I followed the directions on the following website, and that seemed to work for me. Took forever to get everything right to be able to install it. I am not sure if you can do anything with the following links. Hopefully you can and can implement it in to Build-a-pi. I know I would appreciate it. The JTDX website is no longer up and running for the time being. It seems the creator walked away from the project. Hopefully the following will help... 

git clone git.code.sf.net/p/jtdx/code jtdx-code

https://jtdx.freeforums.net/thread/112/deb-installer-raspberry-pi

These gave me the latest build of JTDX. 

SourceForge has the install files and a GIT hub as well

https://sourceforge.net/projects/jtdx/files/jtdx_2.2.159/

I am not sure what you mean about the SDR Software and BAP. Could you explain a little further?

I am curious if your software would work on a 64bit system? I have a Raspberry Pi 4B that I would like to install this on, but was uncertain if it would work...

Keep up the excellent work!!