km4ack/pi-build #514: BAP install failed on VARA

canonball2, 2023-05-27:
I was trying to do a fresh install on Rpi 4 with 64 bit. Ran the main install script just fine. Then it stopped.

WARNING: v3dv is neither a complete nor a conformant Vulkan implementation. Testing use only.

I ran through about 8 lines of the same error for over 2hrs. 
It would not terminate or shutdown or reboot, had to force a shutdown.It rebooted normally but BAP wasn't installed, just had the background graphics to it, no menu items. I tried to rerun the install script again, but it says it's already installed and directories are already present.

Is there a way to re-install or do I have to reimage?

Greg (KE5DXA)

km4ack, 2023-05-27:
This is a known problem at the moment that is being worked on by the Pi-VARA developer. Reference - https://groups.io/g/KM4ACK-Pi/topic/99146340 For now, you will need to skip the VARA install. Everything else should work as normal.

closed 2023-05-27 by km4ack

canonball2, 2023-05-27:
How can I re-install if the script fails to do so. It says the directories already exist and quits. Do just need to delete the BAP directories?

km4ack, 2023-05-28:
Run:
`bash ~/pi-build/update`