km4ack/pi-build #526: Pi-Build out of date?

MistaMx, 2024-03-23:
Howdy hello

Been trying all day to get this setup and working on my Pi4
tried following the text document and then again with the youtube video but seems alot of things are missing or wont install correctly

Started off with the pi-build and got it installed but conky would not start, same with a couple other programs they would give some desktop error.

I can start from scratch for the 29th time and log everything that isnt working correctly following either the youtube or the text only document on setting up the pi build

Question is.. is there a updated version of this? All i want to do is have conky going and be able to run APRS on the pi as a igate/digi
I couldnt use a RTL dongle as the receiver could i? Rather then having to plop a portable (more then likely use a GD77 for that if need be)

jeremybox, 2024-04-18:
https://github.com/km4ack/pi-build/pull/527

tonycanike, 2024-04-18:
Oh it's news to me that Build-a-Pi is deprecated. I just installed and configured Build-a-Pi last week.  

How would anyone know it's deprecated?  There are numerous youtube videos out there about Build-a-Pi. I followed the KM4ACK videos.  None of the youtube videos nor the youtube video descriptions mention that it's deprecated. 

Now that I have Build-a-Pi configured and running, is there an upgrade path from Build-a-Pi to 73Linux?

km4ack, 2024-04-18:
There were later video announcing that 73Linux would replace BAP but I should have updated the readme on GitHub. Running 73Linux on top of BAP is the path to upgrade. There's no need to uninstall anything. After moving to 73Linux, you can safely remove the pi-build directory.

jeremybox, 2024-04-18:
I can confirm this process works (as I did it yesterday).  Thanks Jason!